ros2/design #26: Write ament design document

wjwwood, 2015-03-03:
AC:

- [ ] Whitepaper describing build tool design
  - [ ] appendix describing changes from catkin

jacquelinekay, 2015-07-09:
Dirk started an effort in https://github.com/ros2/design/pull/38. Would you consider this issue closed or are you planning on expanding it?

http://design.ros2.org/articles/ament.html

dirk-thomas, 2015-07-09:
The article does not yet cover the design but focuses on the differences to catkin.

dirk-thomas, 2015-07-27:
With the recent merge of #44 I think ament is described well enough for now. If nobody disagrees I would close this ticket soon.

wjwwood, 2015-07-27:
The only thing that I think we could describe in more detail is how we changed setup files in ament to have local and "recursive" setup files and that we use the tool to create top level setup files to avoid installation collisions where ever possible.

wjwwood, 2015-07-27:
+1 to close with https://github.com/ros2/design/issues/45

closed 2015-07-28 by dirk-thomas