ros2/design #28: Application trying to access public and private repos

stonier, 2015-03-29:
Just thought I'd raise this issue if continuing with this style in the future. When clicking the 'login with Github' from the main [design page](http://design.ros2.org/) it is requesting permission to access public and **private** repos. Probably don't want it to access people's private repos.

```
Modify authorization
ROS2 Design Website by @ros2 would like to modify its permission to access your account
 Added permissions
Repositories
Public and private
This application will be able to read and write all public and private repo data. This includes the following:

Code
Issues
Pull requests
Wikis
Settings
Webhooks and services
Deploy keys
```

wjwwood, 2015-03-29:
Yeah, I was unable to get one of the features of the site to work without the `repo` scope which is where the private access comes from. I can switch it to `read:org`, but there will be some errors in the javascript, see:

https://developer.github.com/v3/oauth/#scopes

wjwwood, 2015-03-29:
I changed it in d6f2f6c, please logout and back in and let me know if you have trouble viewing.

commit d6f2f6c2 (2015-03-29): remove the need for private access

fixes #28
closed 2015-03-29 by wjwwood