ros2/design #29: Style review/clarifications

jacquelinekay, 2015-06-11:
I am in the process of a style review of the design docs, to help myself get up to speed with ROS 2 design. I will use this issue thread for clarification questions and eventually submit a PR with some changes for style and brevity. Please let me know if I should submit small things like this in separate PRs, or if I should use this thread for overall style comments.

@wjwwood, can you clarify the sentence in discovery_and_negotation.md:

"The existence of a node, its host machine and host process are products of the graph's execution, which would be done by the user executing each in turn, or by a process management system."

so the graph's execution "is done by" the user executing each in turn? What are you referring to with "each"?

I would interpret the sentence as the following, please correct me if I'm wrong:

"The node's existence, its host machine, and its host process are side effects of the graph's execution. These details are determined by the user initiating execution of the graph, or by a process management system."

wjwwood, 2015-06-11:
That interpretation is correct (and clearer), but where "user initiating execution of the graph" translates to the user runs the executables manually.

dirk-thomas, 2015-06-11:
I have numerous style fixes for a few articles in my working copy. I will create PRs for them in the afternoon.

dirk-thomas, 2015-07-15:
Are there any more specific todos related to this ticket or can we close this for now?

jacquelinekay, 2015-07-15:
I'm fine with closing it, too focused on other tasks right now to continue
reviewing the design documents.

closed 2015-07-15 by dirk-thomas