ros2/design #37: Compile error on Windows

cognitivesystems, 2015-06-30:
Hi,

I need to get a Intel RealSense App talking to the rest of my ROS compoenents which run on Linux. Therefore, I was trying to compile Ros2 on Windows 8.1 with VS 2015 and Python 3.5 installed. I am stuck at the compilation "python src\ament\ament_tools\scripts\ament.py build" with 2 similare error as follows

============================================================================
 9>C:\Program Files (x86)\MSBuild\Microsoft.Cpp\v4.0\V140\Microsoft.CppComm
   on.targets(170,5): error MSB6006: "cmd.exe" exited with code 1. [C:\dev\
   ros2\build\builtin_interfaces\builtin_interfaces__cpp.vcxproj]
 9>Done Building Project "C:\dev\ros2\build\builtin_interfaces\builtin_inte
   rfaces__cpp.vcxproj" (default targets) -- FAILED.
 5>Done Building Project "C:\dev\ros2\build\builtin_interfaces\builtin_inte
   rfaces__cpp.vcxproj.metaproj" (default targets) -- FAILED.
10>CustomBuild:
     Traceback (most recent call last):
       File "C:/dev/ros2/install/lib/rosidl_generator_c/rosidl_generator_c"
     , line 45, in <module>
         sys.exit(main())
       File "C:/dev/ros2/install/lib/rosidl_generator_c/rosidl_generator_c"
     , line 40, in main
         args.template_dir,
       File "C:\dev\ros2\install\Lib\site-packages\rosidl_generator_c\__ini
     t__.py", line 50, in generate_c
         interpreter = em.Interpreter(
     AttributeError: module 'em' has no attribute 'Interpreter'
10>C:\Program Files (x86)\MSBuild\Microsoft.Cpp\v4.0\V140\Microsoft.CppComm
   on.targets(170,5): error MSB6006: "cmd.exe" exited with code 1. [C:\dev\
   ros2\build\builtin_interfaces\builtin_interfaces__c.vcxproj]

============================================================================

The EmPy modules exists and the doc show that Interpreter class exists.

Any help would be much appreciated.

Regards,

Suraj

cognitivesystems, 2015-06-30:
fixed it by uninstalling teh version installed through pi and manually installing the version 3.3.2 for empy

wjwwood, 2015-06-30:
Hi,

Rhanks for trying out ROS 2, but it really isn't ready for use yet. See: 

https://github.com/ros2/examples/issues/34#issuecomment-115874771

Also, please post issues like this on the ros2/ros2 issue tracker if you don't know where else to put it. Thanks!

closed 2015-06-30 by wjwwood

cognitivesystems, 2015-06-30:
> ros2/examples#34 (comment)

Thanks William!

I got a bit further with compilation but after a while im getting 45 linking errors! difficult to debug on windows especially with limited scroll in shell unlike linux! will continue tomorrow

if u think its not ready I would for the time being have a intermediet middleware like ice or thrift! rosbridge is not an option as I need to host services on the windows side aswell

cognitivesystems, 2015-06-30:
fyi I also tried win_ros on win7 but on win 8.1 the setup instructions dont work anymore