ros2/design #61: design how to support component node migration between containers

tfoote, 2015-09-22:
This was mentioned in #34 but needs it's own discussion. deferred [here](https://github.com/ros2/design/pull/34#issuecomment-141509421)

jacquelinekay, 2015-09-28:
Semantic nitpick:

Could we use a different word than "migration" for this issue? The term is a bit overloaded from the mostly unrelated context of migrating to a new version of an API, etc.

Some other ideas:
"transferring" nodes between containers
"swapping" nodes between containers

I suppose an argument against these choices, especially "swapping", is that they imply that the process is fast and seamless, while "migration" implies that the process takes time and overhead (copying the node resources to a new container).

dirk-thomas, 2015-09-28:
I think the term `migration` fits this use case very well and does by no mean imply effort. The same term is used for very similar technology: https://en.wikipedia.org/wiki/Live_migration