ros2/design #77: rendering of C code samples

codebot, 2016-01-28:
I'm not sure if this is just my browser, but the triple-backtick on [this page](http://design.ros2.org/articles/realtime_background.html) (e.g., the "Memory Management" section) seems to not quite render the following C code correctly. Maybe it needs a triple-tilde instead? I don't know. (the content is awesome, btw... this is just about its rendering)

dirk-thomas, 2016-01-28:
I am not sure why but maybe #78 helps?

jacquelinekay, 2016-01-28:
Yep, I think it's because the code is overindented

dirk-thomas, 2016-01-28:
Doesn't fix it though...

codebot, 2016-01-28:
triple-tilde before and after the code segment causes Jekyll to make a nice
code block, but it doesn't have syntax highlighting

codebot, 2016-01-28:
triple-tilde with "c" after it, like this:

~~~c
printf("hello, world\n");
~~~

seems to turn on syntax highlighting, but also brings in wonky line numbers
without spaces after them.

jacquelinekay, 2016-01-28:
Apparently you can suppress line numbers by editing a `syntax.css` file

https://demisx.github.io/jekyll/2014/01/13/improve-code-highlighting-in-jekyll.html

jacquelinekay, 2016-01-28:
Trying to figure out how to preview a github/Jekyll markdown page based on a branch of the source repository; I think the best way is to set up a local instance of the page based on a branch:

http://jekyllrb.com/docs/github-pages/#project-page-url-structure

codebot, 2016-01-28:
i got it working by `apt-get install jekyll`  then cloning the repo,
checking out the branch you want, and then running this from the checkout
root:

jekyll --server --baseurl ''

then you point firefox to http://localhost:4000 and it works

tfoote, 2016-01-28:
There's a script ./test_local_deploy.sh and it will run the test server inside docker and expose it on port 4000.

jacquelinekay, 2016-01-29:
@tfoote, where do I get the ros_tutorials docker image? That script is failing because docker doesn't know where to find it.

jacquelinekay, 2016-01-29:
Tully helped me out with updating the Dockerfile, but I still can't view the instance hosted by docker on localhost:4000. I think there's something screwy happening with my iptables rules.

tfoote, 2016-02-02:
Suggest just giving up on syntax highlighting for now.

codebot, 2016-02-03:
#80 will hopefully just render as non-highlighted text. But we'll have to run it through the correct version of the Jekyll pipeline first to make sure it looks OK.

dirk-thomas, 2016-02-03:
No need to validate #80 - generic code blocks do work.

closed 2016-02-03 by dirk-thomas