ros2/design #85: ROS Microservices

omidkrad, 2016-03-15:
Hi, would it be possible to create [Microservices](http://microservices.io/patterns/microservices.html) (for not necessarily robotics applications) using ROS 2? If yes, it would be great if you could provide some packages/resources for that along the way so that those who are interested can grab just the core run-time of the framework that is needed for creating their business microservices without the robotics components of it. I think it would be a very valid use case to use ROS as the abstraction layer of DDS transports. Thanks!

dirk-thomas, 2016-08-05:
I am sorry for the late reply.

The ROS 2 core communication feature and the client libraries for different languages make no specific assumptions about the context they are being used in. The same I would argue for ROS 1. So it should be perfectly feasible to use either of them for non-robotics applications.

I will mark this ticket as closed for now since it is unclear to me what action we should take for it. If there are any aspects you are aware of which should be changed to make it fit your use cases please feel free to comment here and the ticket can be reopened.

closed 2016-08-05 by dirk-thomas

sebastienmascha, 2020-10-18:
> Hi, would it be possible to create [Microservices](http://microservices.io/patterns/microservices.html) (for not necessarily robotics applications) using ROS 2? If yes, it would be great if you [...]

Any example of architecture that you have in mind or implemented @omidkrad ?