ros2/design #96: finish EBNF for topic and service names

wjwwood, 2016-10-29:
See: https://github.com/ros2/design/pull/90#issuecomment-247329344

Theosakamg, 2016-11-11:
In design article it say :

> must not have an underscore (_) followed by a forward slash (/), i.e. _/

https://github.com/ros2/design/blob/gh-pages/articles/115_topic_and_service_name_mapping.md#ros-2-topic-and-service-name-constraints

but in valid example :

> foo/_/bar

https://github.com/ros2/design/blob/gh-pages/articles/115_topic_and_service_name_mapping.md#ros-2-name-examples

Can you clarify ?

Theosakamg, 2016-11-11:
ind design acticle it say :

> must not contain any number of repeated underscores (_)

https://github.com/ros2/design/blob/gh-pages/articles/115_topic_and_service_name_mapping.md#ros-2-topic-and-service-name-constraints

but in current implementation of Parameter service :
`node_->get_name() + "__get_parameters",`
https://github.com/ros2/rclcpp/blob/5894a9cd4eb0371e380dcbac73b74efae353fde2/rclcpp/src/rclcpp/parameter_service.cpp

It is excluded of Service name of parameter stack ?

wjwwood, 2016-11-11:
@Theosakamg looks like I missed that one (`foo/_/bar`) when updating the rules. Thanks for pointing it out! I'll open a pr to fix it.

As for the current implementation, it doesn't follow this design doc yet (the implementation is lagging behind the document). We'll replace that line with something like this:

```c++
node_->get_name() + "/_get_parameters",
```

Or something similar, so that it uses the `/` namespace separator and is "hidden".

If you have further comments, please open a new issue or comment on the original pr (https://github.com/ros2/design/pull/90). This issue is specifically about completing the EBNF (candidate implementation is in #96).