ros2/design #124: Forward slash in DDS topic name in example

miguelprada, 2017-04-03:
The _Examples and Ideas for Communicating with Non-ROS Topics_ section in _Topic and Service name mapping to DDS_ article presents an example where a DDS topic name containing a forward slash is discussed.

Although the article discusses an ambiguous specification of an "exact" topic name, I have the impresion that one of the two interpretations should not be legal in DDS. Specifically I'm referring to the mapping from `rostopic+exact://camera_left/image` to DDS topic `camera_left/image` and partition [].

Just bringing it up since it can lead to confusion. It probably makes little sense to combine `rostopic+exact` with a name containing forward slashes, right?

wjwwood, 2017-04-03:
@miguelprada thanks for pointing it out. I think the purpose of that last example was to bring up that issue with the concept of `rostopic+exact://`, which I thought would be addressed by the next line:

> Considering the third case, it’s not clear if the `/` should be respected or if additional syntax is required to support partitions in the case of “exact” topic names.

Perhaps it should just be made clearer that this last case is intentionally problematic.

Karsten1987, 2017-04-04:
From what I understand in the [linked](http://community.rti.com/rti-doc/510/ndds/doc/html/api_cpp/group__DDSQueryAndFilterSyntaxModule.html) spec, DDS topics don't exclude a forward slash. AFAIS, it only has a special meaning within the SQL Extension of rti. 
With that in mind, you could imagine a ROS2 independent DDS topic with a forward slash. If you want to subscribe to such a topic, you may not be able to apply the same topic rules as for ROS2. 

But I have to do some more research on that.

wjwwood, 2017-04-04:
DDS topics only allow alphanumeric and underscores (and must not start with a number) according to the spec:

> In DDS, topic names are restricted by these restrictions:
>
> *From DDS 1.4 specification, or the [RTI documentation](http://community.rti.com/rti-doc/510/ndds/doc/html/api_cpp/group__DDSQueryAndFilterSyntaxModule.html):*
>
> `TOPICNAME - A topic name is an identifier for a topic, and is defined as any series of characters 'a', ..., 'z', 'A', ..., 'Z', '0', ..., '9', '_' but may not start with a digit.
Note: that the DDS specification has a known typo, where it says - are allowed, but the RTI documentation correctly lists _ as allowed.`
>
> Additionally, DDS - or more precisely the underlying RTPS protocol - has a hard limit on topic names of 256 characters, so an additional goal is to minimize the number of extra characters used when mapping from ROS to DDS names. See The Real-time Publish-Subscribe Protocol (RTPS) DDS Interoperability Wire Protocol Specification, Table 9.12 for more details.

-- http://design.ros2.org/articles/topic_and_service_names.html#dds-topic-names

@Karsten1987 What makes you think forward slash is allowed in a topic name? It might be allowed by the implementation, but we have to go by the specification since we're not tied to one implementation.

Karsten1987, 2017-04-04:
I totally agree with what you said. We will comply to the specification, however, as you already mentioned, the implementations may be relaxed on these and you technically could end up with one forward slash in a DDS topic. If so, the question would be if we excluded that topic from being available in ROS.

If we decide to not allow the forward slash, then I like the proposed rephrasing of this PR.

wjwwood, 2017-04-04:
I see what you mean, we can let the user pass a `/` into a DDS topic name by using "exact", and if the implementation doesn't allow it, let it raise. Basically "exact" would circumvent all of our checking.