ros2/design #188: rclcpp::init must be called before everything?

reed-lau, 2018-08-23:
Recently, I find that we must call rclcpp::init (or rclpy::init in python) before everything. But in some cases, I wana just make a node and create some publishers and subscribers for later usage. For example, i would like put the node( with some publishers and subscribers) into a library. but the init make it hard todo that. In ros1 it easy to do that. Do i have some workaround to achieve it.

wjwwood, 2018-08-23:
No, from a conceptual stand point, you must call init before creating a Node.

You should have your users (the users of your library) call init before using your library or, if you're not sure if they're going to use ROS or not, then you could check if init has been done and do it yourself if not (relies on the feature request I linked to above). There is already an issue on `rclpy` for checking if init has been called: https://github.com/ros2/rclpy/issues/190

Another future option would be to have the ability to init multiple times in the same process (non-global init), which is a refactor we've been wanting to do for a while now. But that's not an option right now.

wjwwood, 2018-08-23:
Please close this issue if that answers your question and/or consider closing this in favor of feature requests on one of the client libraries (`rclpy` or `rclcpp`).

reed-lau, 2018-08-23:
for multiple rclpy.init() call and init fail will both throw RuntimeError. the right way to avoid multiple init is like that ?
```
try:
    rclpy.init()
execpt Execption:
    if rclpy.ok():
       print("multiple init")
    else:
       printf("init fail")
```

closed 2018-08-29 by reed-lau