ros2/design #251: Intra-Process Communications for all language clients

emersonknapp, 2019-08-23:
## Description
This issue is a call for a design of zero-copy intra-process communications available to all ROS2 language clients.

The current implementation of this feature exists in `rclcpp` and therefore is not usable for Python (or less-supported languages C, Java, Rust, etc.)

## Acceptance Criteria
To close this issue, we want a design document that proposes the architecture for
* Intra-process optimized communications
* zero copy
* Available to all ROS2 language clients (therefore exists at or below the level of `rcl`)

As a follow up, will attempt to collect existing thoughts from https://github.com/ros2/design/pull/239 and add as comments below.

## Note
I do not consider myself personally an expert on this, however I'm very interested in collaborating towards a top down view of what this part of the ROS2 core should look like, and figuring out how the community can pull together towards a solution

ivanpauno, 2019-08-26:
I think that we can take some ideas from Connext "zero copy transfer over shared memory":
- https://community.rti.com/static/documentation/connext-micro/3.0.0/doc/html/usersmanual/zerocopy.html
- https://community.rti.com/static/documentation/connext-dds/current/doc/manuals/connext_dds/html_files/RTI_ConnextDDS_CoreLibraries_UsersManual/Content/UsersManual/SendingLDZeroCopy.htm

That's actually interprocess communication over shared memory, but something similar can be replicated using a buffer instead of a piece of shared memory.

The basic idea is that you have to ask to the publisher for a new message, instead of allocating an `unique_ptr`:
```cpp
msg = publisher->get_new_message();
if (msg != nullptr) {
  msg->data = "asd";
  publisher->publish(msg);
}
```

Currently, message lifetime can be extended to be longer than the scope of the callback (in cpp). That would not be possible if we go ahead with something like this (or at least, it will be really hard to implement that feature).

allenh1, 2019-08-26:
@ivanpauno I don't think `publisher->get_new_message()` ever return `nullptr`. I'd prefer a more asynchronous way to fetch a message, or potentially blocking on that call instead. I'm not very fond of the blocking call idea, but maybe an asynchronous trigger could be set up?

Maybe it could be set up so that we can `std::invoke` a callback in the `publish()` function? This isn't great though, since this would need to be done in `rcl`, which means it would be wasting cycles checking if there are `std::bind`ed callbacks on non-shared memory platforms.

I'm not seeing a way to make this happen in anything above `rmw`, except of course when there are multiple nodes inside the same process.

Sorry for the rambles, very interested in this idea.

fujitatomoya, 2019-08-27:
just sharing my thought,

> The implementation could live in rcl or rmw, I'm not sure what would be better.

i believe that it is better to be implemented in rmw, not rcl.

- it sounds rmw responsibility to take care of transportation. ([rmw](https://index.ros.org//doc/ros2/Concepts/DDS-and-ROS-middleware-implementations/))
- provide consistent/compatible API to frontend, concealed by rmw.
- taking advantage/comparison of each rmw implementation.

qootec, 2019-09-17:
I initially posted this as an topic on answers.ros.org (see https://answers.ros.org/question/333180/ros2-micro-ros-intra-process/) but was advised by the moderator to move it to discourse...  I think the core of my concern touches your discussion.

(My context: ROS2 inside a machine controller)

Looking at your proposals for intra-process communication, I fail to see whether you also take into account the multi-priority requirements such (often embedded) environments typically have.

I currently see fragmented solution elements or approaches:

* From Micro-ROS: Multiple executors could be hosted in the same process/node, each having their own queue for messages (or in fact their handlers) of the corresponding priority (based on their handlers' callbackgroup priority).

* From ROS2: ROS2 does not create its own queuing mechanism, but instead relies on the queues already available in the DDS middleware.

* From ROS2 (close to this topic): use_intra_process_comms() … if true, messages will go through a special intra-process communication code path. So potentially excluding DDS. Then how will they get queued / priority managed?

* (RTI) DDS has a Transport_Priority_QoS defined per DataWriter, which is then to be kept in sync with the cbGroup priority?

Is there any documented vision on how your intra-process-communication would co-exist with multi-priority queuing/handling? 

Johan

gavanderhoorn, 2019-09-17:
> I initially posted this as an topic on answers.ros.org (see https://answers.ros.org/question/333180/ros2-micro-ros-intra-process/) but was advised by the moderator to move it to discourse... 

I did, but this is not the embedded category on ROS Discourse.

atyshka, 2020-08-12:
Any updates on this roughly a year later?

ivanpauno, 2020-08-13:
> Any updates on this roughly a year later?

Not that I know of.
The problem isn't trivial, and AFAIK there is no people assigned to work on it.

twaddellberkeley, 2021-09-11:
Hi @ivanpauno, is there any work on this problem, if not do you need help? Would love to dive into it.

Cheers

ivanpauno, 2021-09-13:
AFAIK, nobody is working on this right now.
I'm not sure if there's a plan to work on the topic soon.

emersonknapp, 2021-09-13:
I'm not sure, but does the Cyclone+iceoryx combo do this automatically for C++ nodes in the same process?

ivanpauno, 2021-09-13:
> I'm not sure, but does the Cyclone+iceoryx combo do this automatically for C++ nodes in the same process?

Not zero copy, zero-copy requires a different API.