ros2/design #265: Middleware alternatives to DDS

infiBoy, 2019-11-15:
Can you survey the alternatives that you considered to DDS?

dirk-thomas, 2019-11-15:
Please see the following design documents:
* http://design.ros2.org/articles/why_ros2.html
* http://design.ros2.org/articles/ros_on_dds.html
* http://design.ros2.org/articles/ros_middleware_interface.html

While the officially supported RMW implementations use DDS there are several non-DDS implementations implementing the RMW interface.

closed 2019-11-15 by dirk-thomas

infiBoy, 2019-11-15:
Thanks for the answer, but I looked forward to the answer of **which** is the list of " non-DDS implementations implementing the RMW interface."? ... and didn't find it by your links.

dirk-thomas, 2019-11-16:
> list of " non-DDS implementations implementing the RMW interface."

RMW repos of non-DDS implementations in the `ros2` org unit on GitHub:
* https://github.com/ros2/rmw_dps
* https://github.com/ros2/rmw_iceoryx

Other implementations announced on Discourse:
*OPC UA and NDN: https://discourse.ros.org/t/non-dds-based-rmw-implementation/5890/3?u=dirk-thomas
* NDP: https://discourse.ros.org/t/ros2-on-microcontrollers-with-riot/1293/9

And probably others I am currently forgetting or I have never heard about before.

ZhenshengLee, 2023-10-26:
also there is https://github.com/eclipse-ecal/rmw_ecal