ros2/design #280: Add design document on configuring QoS at runtime

jacobperron, 2020-04-22:
It is not uncommon to have a node provide a topic publisher or subscription that could be have different valid QoS settings depending on the application. In these situations, we see node authors offering ways to configure the QoS (typically in the form of a ROS parameter). For example,

https://github.com/ros-perception/image_pipeline/pull/521
https://github.com/SteveMacenski/ros2_ouster_drivers/pull/26
https://github.com/ros-simulation/gazebo_ros_pkgs/issues/1079

I think it would be valuable to have a common mechanism for configuring QoS at runtime. Having a consistent interface for node authors and integrators to use would hopefully make their lives easier.

See https://github.com/ros-simulation/gazebo_ros_pkgs/issues/1079 for some early discussion.

gbiggs, 2020-07-13:
This is possibly something that can be done via a set of well-known parameters.

ivanpauno, 2020-07-28:
I've renamed the issue from `runtime` to `startup time`.
Re-configuring QoS profiles at runtime is much harder, and not what the original issue was aiming for (the three linked workarounds are for startup qos configuration).

The idea is to have a way to externally configure the QoS settings when reusing a node someone else has written.
For different use cases, different QoS profiles may be desired.

This also matches what DDS allows.

jacobperron, 2021-01-13:
A design doc was added in https://github.com/ros2/design/pull/300.
And subsequently implemented in https://github.com/ros2/rclcpp/pull/1408 and https://github.com/ros2/rclpy/pull/635

closed 2021-01-13 by jacobperron