ros2/design #285: Update XML schema definition for launch files

joshwapohlmann, 2020-06-12:
## Problem
The only XML schema definition for launch files I could find was [in the design docs](gh-pages/articles/151_roslaunch_xml.md). 
But it is outdated, since e.g. it expects the `<node>` tag to have a `executable` attribute, but it is actually `exec`.
Now, I don't know about other inconsistencies, but I want to use this in my launch files. So what do do?

## Proposed Solution
1. Update the [schema definition](https://github.com/ros2/design/blob/gh-pages/articles/specs/launch.0.1.1.xsd)
2. Host a copy on a public url

1. Provide it as part of the [launch repository](https://github.com/ros2/launch)
2. Delete the outdated schema definition and link to the file in the repository.

## Related
https://github.com/ros2/launch/issues/392

ivanpauno, 2020-06-26:
Closing, as https://github.com/ros2/design/pull/286 was merged.
@joshwapohlmann If there's any other necessary update in the schema please let me know and I will help to fix it.

ivanpauno, 2020-06-26:
> Host a copy on a public url as proposed in ros2/launch#392

I will reply about that in #392.

closed 2020-06-26 by ivanpauno