ros2/design #288: Article numbering is not clear

ivanpauno, 2020-06-26:
Some of the [articles](https://github.com/ros2/design/tree/gh-pages/articles) start with an article number in the name, some others don't.
Some articles have a repeated number (`150`).

It's not clear how article numbers are assigned and if an article number is required.
I haven't found documentation clarifying how the numbering is decided.

dirk-thomas, 2020-07-16:
The articles are listed in the sort order of their filenames. The numbers are used to make them appear in a specific order - often with gaps between numbers to allow inserting new articles in between. Some articles don't have numbers since the author probably didn't intend the article to have a specific position so they appear after the numbered articles in alphabetical order.

@ivanpauno Please either close this question if this answer it and you don't think there is a need for follow up work or decribe what you think should happen / change.

ivanpauno, 2020-07-16:
I propose the following changes:

- 150_ros_command_line_arguments.md -> 160_ros_command_line_arguments.md (there's already an article numbered 150)
- 160_roslaunch_frontend.md -> 152_roslaunch_frontend.md (all launch related articles seem to be 15*)
- 142_idl.md -> 115_idl.md (all 11* are about idl, the others 14* articles are about remapping)
- 143_legacy_interface_definition.md -> 116_legacy_interface_definition.md (same as previous item)
- WideStrings.md -> 117_wide_strings.md (seems idl related too)

I think that solves all current inconsistencies. There will still be unnumbered articles, but I'm fine with that.

dirk-thomas, 2020-07-16:
Please go ahead and create a pull request for the proposed changes.

closed 2020-07-17 by ivanpauno