ros2/design #298: zero-copy: shared memory using external mapped buffer

Qzhaohong, 2020-09-16:
We want to transmit camera images to ros2. And actually camera buffers have already allocated in external by camera frameworks / driver. 

I checked the zero-copy guide https://github.com/ros2/design/blob/gh-pages/articles/zero_copy.md. It says above situation is of "Additional Publisher Use Case", and "it will be out of scope for this document, but it is mentioned for completion". So what's the meaning of "mentioned for completion" ? Is it under development or has it already supported?

Thanks

fujitatomoya, 2020-09-16:
I think that is the use case for [additional-publisher-use-case](https://github.com/ros2/design/blob/gh-pages/articles/zero_copy.md#additional-publisher-use-case).

> Is it under development or has it already supported?

but AFAIK, negative on this.

Qzhaohong, 2020-09-16:
Thanks @fujitatomoya 

> I think that is the use case for [additional-publisher-use-case](https://github.com/ros2/design/blob/gh-pages/articles/zero_copy.md#additional-publisher-use-case).
> 
> > Is it under development or has it already supported?
> 
> but AFAIK, negative on this.

wjwwood, 2020-10-01:
> So what's the meaning of "mentioned for completion"?

We mentioned it because it's related to zero-copy functionality, but it's out of scope because we were not planning implement it right now.

It would require additional API to indicate to the middleware that the memory being given is safe to use without copying, and a way for the middleware to notify the user (or for the user to poll if the middleware is still using the memory), so that the user knows when they can reuse or free the memory. We didn't want to take the time to define those APIs and describe their behavior in the design document you linked to. It would require additional work to get there.

> Is it under development or has it already supported?

We are not currently planning to implement this or even update the design document to incorporate it, but you or anyone else can propose changes.

---

I'm going to close this because I believe the issue has been addressed. Feel free to continue to comment and I can reopen if necessary.

closed 2020-10-01 by wjwwood