ros2/design #313: is meta-message transfered via DDS?

Aerotic, 2021-02-28:
According to the following text from[https://github.com/ros2/design/blob/gh-pages/articles/intraprocess_communication.md](https://github.com/ros2/design/blob/gh-pages/articles/intraprocess_communication.md):

>  In order to extract a message from the IntraProcessManager two pieces of information are needed: the id of the Publisher (in order to select the correct ring buffer) and the position of the message within its ring buffer. A meta-message with this information is created and sent through the ROS 2 middleware to all the Subscriptions, which can then retrieve the original message from the IntraProcessManager

It seems meaning that when using intra-process communication, a meta-message must be sent to `IntraProcessManager` every publish via FastRTPS(or another DDS middleware). But I didn't detect any activity of DDS after initialization phase. Besides, I find that the arguments like `publisher_id` transfered among functions may be the so called `meta-message`.

In addition, the following description seems to disaccord the code (<ROS2 FOXY FULL BUILD>/src/ros2/rclcpp/rclcpp/include/rclcpp/experimental/subscription_intra_process.hpp), There is an instance of `BufferUniquePtr` in `SubscriptionIntraProcess` and I think `BufferUniquePtr` is the "ring buffer".  
> Here the message will be stored in the ring buffer associated with the Publisher.

mauropasse, 2021-03-05:
Before the `IntraProcessManager` were created, there was a proper meta-message being sent through the DDS with some info needed to retreive the intra-process message. The CPU performances were good for really big messages, but when small messages were used (10b) the need for a meta-message resulted in poor performances, because sending a message through the DDS is not cheap.
Now everything is handled by the `IntraProcessManager`, no message is sent through the DDS.
But, there's still some DDS activity. The publisher publishes an intra-process message an triggers the subscription DDS guard condition. This happens and goes through the DDS, so the DDS is not really bypassed when using intra-process.

Aerotic, 2021-03-06:
> But, there's still some DDS activity. The publisher publishes an intra-process message an triggers the subscription DDS guard condition. This happens and goes through the DDS, so the DDS is not really bypassed when using intra-process.
Sorry, I've found something different from the above statement. At `src/ros2/rmw_fastrtps/rmw_fastrtps_shared_cpp/src/types/guard_condition.hpp:46`,  it seems that what the guard condition triggering process mainly does is `conditionVariable_->notify_one()`. So this may be not doing with DDS like eProsima FastRTPS

mauropasse, 2021-03-08:
Yes, you're correct. Is the RMW guard condition, not DDS. What I meant was that there is some RMW specific work after all, not all communication stays in **rclcpp**. The goal with IPC (rclcpp, not DDS) was totally by-pass the specific RMW's for passing messages.

clalancette, 2021-03-25:
Since this is a question, and there is no further activity, I'm going to close this out.

closed 2021-03-25 by clalancette