ros2/design #317: Update Launch XML Schema

rob-clarke, 2021-06-24:
Similar to #285 I believe the schema doesn't match the current implementation.

In the schema, the `<node>` element currently defines a `ns` attribute but this appears to have been renamed to `namespace`. At least if [here](https://github.com/ros2/launch_ros/blob/58334b4447daf1d86aa7e17bb5e1b9ebb74301b4/launch_ros/launch_ros/actions/node.py#L319) is where it is being parsed.

@ivanpauno Mentioning you as it's similar to the other issue you dealt with.

ivanpauno, 2021-06-29:
@rob-clarke you're correct, the correct attribute is `namespace` and not `ns`.
A PR is welcomed.

> defines a ns attribute but this appears to have been renamed to namespace

The attribut has always been `namespace`, but it was wrongly documented as `ns` in the schema.

christophebedard, 2021-06-30:
I can take care of this (if @rob-clarke isn't already on it) and also add `ros_args` (https://github.com/ros2/launch_ros/pull/253) at the same time

christophebedard, 2021-06-30:
> I can take care of this (if @rob-clarke isn't already on it) and also add `ros_args` ([ros2/launch_ros#253](https://github.com/ros2/launch_ros/pull/253)) at the same time

I just went ahead and did it: #318

PR #318 (2021-06-30): Update launch XML schema to rename <node> attribute ns to namespace and add ros_args
Closes #317

This renames the `<node>` `ns` attribute to `namespace` (#317). This also adds a new `ros_args` attribute, see https://github.com/ros2/launch_ros/pull/253.
closed 2021-06-30 by ivanpauno