ros2/design #333: Field names of messages and services must be lowercase

Maverobot, 2023-02-17:
In the design draft, it says

> Field names must be lowercase alphanumeric characters with underscores for separating words. They must start with an alphabetic character, they must not end with an underscore and never have two consecutive underscores.

Capital letters in field names can sometimes more suitable and human-readable especially when it comes to mathematical expressions, e.g. in [FrankaState](http://docs.ros.org/en/kinetic/api/franka_msgs/html/msg/FrankaState.html). Are there special reasons why capital letters must be eliminated from the field names?

clalancette, 2023-03-02:
The original design article went in as part of #32 .  There was no rationale given for that particular rule there.

The wording of "must" in that article is in [RFC2119](https://www.rfc-editor.org/rfc/rfc2119) style, which is why it is very imperative.

In short, we'd be OK with lifting this restriction.  However, we do want to keep the core using only snake_case fields (for consistency), which we would want to do with a linter.  Thus, in order to lift this restriction the following would need to be done:

1. Develop a linter for .msg/.srv/.action/.idl files that enforces some rules, and optionally enforces others.
2. Add something to ament_lint to enable that linter to be easily called from ament.
3. Update the .msg/.srv/.action/.idl parsers to allow uppercase letters.
4. Update https://github.com/ros2/design/blob/gh-pages/articles/116_legacy_interface_definition.md and https://github.com/ros2/design/blob/gh-pages/articles/110_interface_definition.md to allow that style.

We aren't likely to have time to do this, so if you'd like to contribute, we'd be happy to review PRs to do the above.

gavanderhoorn, 2023-03-05:
I've always understood the main rationale for the no-capitals rule-of-thumb to be the fact it significantly reduces possibility for errors and makes 'guessing' the exact spelling of fieldnames much easier.

My experience with other middlewares/code generation frameworks is that, despite the fact we have extensive code-completion support these days, no-capitals is *very* convenient. Personally, I believe it outweighs the fact fieldnames can't follow naming of things like mathematical expressions, but this is just one voice/data point of course.

clalancette, 2023-03-06:
Thanks for the historical notes @mikaelarguedas !