ros2/design #336: where is the discauss of register callback and run callback?

fengmao31, 2023-10-14:
I remember I have read the article about ros client's register callback and action callback. It talks about the how to register the callback. Also, it talk about use poller to get callbacks and run or trigger to run callbacks.

fujitatomoya, 2023-10-14:
can you take a look at https://discourse.ros.org/t/where-is-the-discauss-of-register-callback-and-run-callback/34136.

i think it is not recommended to post the same questions on multiple site...

closed 2023-10-14 by clalancette

fengmao31, 2023-10-15:
I donnot means to find tutorial to know how to use the ros client. That article is about how to and why ros use that way of callback desgin. I use ros for many years. I forget where is the article.

fengmao31, 2023-10-15:
I already ask questions to https://robotics.stackexchange.com/. Thank you for your answer.